MajicDesigns/MD_MAX72XX #1: tried it on the ESP8266

mherweg, 2018-01-14:
thank you for your nice Library and the "daft punk helmet" demo!!!
I did run it on the ESP8266 and it works.

then I tried to add the Wifi & MQTT features that I sue in other projects:
#include <ESP8266WiFi.h>
#include <WiFiUdp.h>
#include <PubSubClient.h>

and it crashes :-(

did someone had more success with that ?

to be more precise:
I got WiFi connected
MQTT ...connected
and it crashes at:

resetMatrix();

MajicDesigns, 2018-01-15:
Crashes or just restarts again? Have you checked if the watchdog timer is triggering this behaviour?

mherweg, 2018-01-15:
restarts over and over.  " initialise the MD_MAX72xx first" ... ok I will try that.

MajicDesigns, 2018-01-15:
You may also want to try some of the watchdog timer resets mentioned here while you are testing.
https://techtutorialsx.com/2017/01/21/esp8266-watchdog-functions/

MajicDesigns, 2018-02-19:
Closed due to lack of activity.

closed 2018-02-19 by MajicDesigns