MajicDesigns/MD_MAX72XX #3: Add to PlatformIO registry?

bkhofmann, 2018-02-19:
Is it possible to get this library added to the PlatformIO Registry like some of your others?

MajicDesigns, 2018-02-19:
As far as I know, this is the ONLY library that should be on PlatformIO because a forked version was on but not updating version. Point me to the registration process and I'll investigate.

bkhofmann, 2018-02-19:
[http://docs.platformio.org/en/latest/userguide/lib/cmd_register.html](url)

Appears it may need to be done within the PlatformIO software.  I'm pretty new to this!

bkhofmann, 2018-02-19:
[http://docs.platformio.org/en/latest/librarymanager/creating.html](url)

Here's another link as well.

MajicDesigns, 2018-02-20:
You are correct that PlatformIO registration requires the use of PlatformIO. As I do not use PLatformIO (nor do I intent to start at this stage), this is not going to happen by me. It does not stop others registering the MD_* libraries if they want to, as the _library.properties_ manifest file is sufficient for PlatformIO to register the libraries.

closed 2018-02-20 by MajicDesigns