MajicDesigns/MD_MAX72XX #4: Issue when using >32 devices

LEdgeley, 2018-02-23:
I have a string of 40 MAX7219 controlled matrices which I'm attempting to control using an Arduino Mega 2560 R3 and the MD_MAX72XX library.

I've used the print_test sample code to get my matrices to display a string of characters ("0123456789ABCDEFGHIJKLMNOPQRSTUVWXYZabcdesfghijklmnopqrstuvwxyz"). Everything works correctly as I change MAX_DEVICES to values in the range 1-32 (as expected, the sequence repeats when the physical string of matrices is longer than MAX_DEVICES):
[image]

However, when I change MAX_DEVICES to a value >32, the subsequent matrices are not controlled.

MAX_DEVICES = 33:
[image]

MAX_DEVICES = 36:

[image]

MAX_DEVICES = 40: 

[image]

MajicDesigns, 2018-02-23:
> the sequence repeats when the physical string of matrices is longer than MAX_DEVICES

This is true only if you send more than once, but I get what you are saying.

Can you please provide more information:
1. How are these matrices wired?
2. What is the power supply (I assume external, but please confirm).
3. Please post the code you are using.

MajicDesigns, 2018-02-24:
Before we go too much further, can you try the latest upload of the library. The column number parameter to setColumn() was uint8_t instead of uint16_t. 32 modules just happens to be 256 columns, so would  overflow the uint8_t. Let me know the result.

LEdgeley, 2018-02-26:
Just tried the new version, and it wouldn't compile - missing a semicolon on line 315 of MD_MAX72xx.h.

However, once I added in the missing semicolon the new version works perfectly on 40 devices. Sadly I don't have 8192 modules to test the new upper bound!

Thanks so much for your fix.

closed 2018-02-26 by LEdgeley