MajicDesigns/MD_MAX72XX #5: string specular and polarity reverse per char

intel2004, 2018-03-30:
i try your library in arduino nano,all the exaple sketch look the same:
every single chars is reversed and in specular position.an imagine can explain here is print_text example:

[image]

MajicDesigns, 2018-03-30:
Did you change the MD_MAX72xx library to tell it which hardware you are using? Please follow the instructions in the library documentation if you have not already done so?

intel2004, 2018-03-30:
i found it:
change MD_MAX72xx_lib.h , line 483 like that:

#if USE_PAROLA_HW   // tested MC 8 March 2014
//#pragma message "PAROLA HW selected"
#define HW_DIG_ROWS 1 ///< MAX72xx digits are mapped to rows in on the matrix
#define HW_REV_COLS 0 ///< Normal orientation is col 0 on the right. Set to 1 if reversed
#define HW_REV_ROWS 0 ///< Normal orientation is row 0 at the top. Set to 1 if reversed
#endif

MajicDesigns, 2018-03-30:
Actually the instructions say to change the defines in the MD_MAX72xx.h file.

closed 2018-03-30 by MajicDesigns