MajicDesigns/MD_MAX72XX #7: Mirrored text

003random, 2018-07-14:
Hey there @MajicDesigns,

When testing the example scripts, the text displays in a wrong order.
When googling for this issue i found that you have to change 
`#define USE_PAROLA_HW   1`
To
`#define USE_PAROLA_HW   0`
but the problem is that the line you have to edit is not present in the file. (MD_MAX72xx.h).
How do i fix this? or is there something im missing?
Thanks in advance :)

003random,

MajicDesigns, 2018-07-15:
From version 3 onwards you specify the hardware type in the code as one of the class parameters. Please read the documentation or look at the examples for how this is done.

closed 2018-07-15 by MajicDesigns

003random, 2018-07-15:
Thanks :)
Great library btw!

AleksandarTheMachine, 2023-03-02:
I have exactly the same problem! What to do?

MajicDesigns, 2023-03-02:
Change the hardware types until it works. Make sure that the wires come into the display at the right hand side of the display.
[https://arduinoplusplus.wordpress.com/2020/08/29/parola-a-to-z-frequently-asked-issues/](url)