MajicDesigns/MD_MAX72XX #12: Rx/Tx Serial input for control of LED Banner

mattvara, 2018-11-02:
Hello MD Parola,  I have been using your MD_MAX72xx Message Serial successfully to control an LED banner from the Arduino suite Serial Monitor, i'm wanting to do the same via the Rx/Tx pins on the board.  I have RS-232 converted to TTL and would like the LED banner to display the ASCii characters I send via my PLC controller to RS-232 output.  I found the below link on the Arduino side and suspect I have to include the same serial library, specify which serial port to use in your code, and then maybe make some other minor modifications.  

https://www.arduino.cc/en/Reference/SoftwareSerialRead

Do you have any suggestions on how to proceed or do you have any examples you could share?

MajicDesigns, 2018-11-03:
The ide monitor is connected to the serial pins so I am not sure what you are trying to achieve by using SoftwareSerial? In any case any of the serial examples will work with the substitution for the SoftwareSerial object instead of Serial.

mattvara, 2018-11-05:
Thanks for the response.   I'm trying to use pins 0 & 1 of the Arduino-Uno to  receive an RS-232 to TTL ASCii data transmission for display to the LED Banner.  I have the serial communication from the ide monitor working, but i'm trying not to use that and read pins 0 and 1 instead.

mattvara, 2018-11-05:
Are you saying that I should have to change your code at all and just disconnect the USB cable from my PC and the board will start communicating via Pins 0, and 1?

mattvara, 2018-11-06:
Great, thanks for your help.  It works.  My RS-232 to TTL converter had already converted the Tx to Rx and Rx to Tx internally so I just had to connect the converters Tx to the Tx pin on the Arduino and ditto on the Rx.

closed 2018-11-06 by mattvara