MajicDesigns/MD_MAX72XX #13: Printing Swe characters

bortek, 2019-02-02:
Apologies before I start writing this question. Might sound trivial but I am looking a simple way to print öäåÖÄÅ (swedish) characters on 8x8 LED (MAX7219) .

Let's say if I take MD_MAX72XX_PrintText from examples and replace output text with this

char message[BUF_SIZE] = {"HelloÅ"};

then I get a "Hello |-" instead of HelloÅ

How in an easy can this be achieved ?

bortek, 2019-02-03:
I looked at the example you are referring to and tried to run it but still could not get umlaut characters to display. 

I was thinking more that there should have been a support in the underlying libraries so that when typing a text with umlauts in the code it should be displayed on the matrix.

MajicDesigns, 2019-02-03:
Always welcome contribution from users. If you extend the libraries in that way for your own use please submit it for review and inclusion in a future release.

closed 2019-02-03 by MajicDesigns

bortek, 2019-02-03:
thanks. Will do.