MajicDesigns/MD_MAX72XX #14: broken links

ieee488, 2019-04-11:
I don't pretend to know how Github works

The link to MD_MAX72xx.h  on https://majicdesigns.github.io/MD_MAX72XX/page_software.html  is broken

All the links on https://majicdesigns.github.io/MD_MAX72XX/files.html  are broken

All the links on https://majicdesigns.github.io/MD_MAX72XX/globals.html   are broken

.

MajicDesigns, 2019-04-11:
If you have the library installed you should have a duplicate of this doc in the MD_MAX72XX/docs folder (open index.html). Local version works on my system - does it work locally for you?

ieee488, 2019-04-11:
Yes, the local version work.

The online version is broken.

MajicDesigns, 2019-04-11:
Looks like this is a problem broken for every other repository as well. I'll note it but can't understand what to do about it for the moment.

closed 2019-04-11 by MajicDesigns