MajicDesigns/MD_MAX72XX #15: FC-16 boards and 1088B matrices

ieee488, 2019-04-14:
I don't understand how to use 1088B LED matrices with FC-16 boards with this library.
What do I need to change in the library?

When using the 1088A, I have to insert the LED matrices upside down.

MajicDesigns, 2019-04-14:
The 1088B are a different type of hardware. You will need to wire them to the MAX7219 IC in a different way from the 1088A. The library won't need to change, but you may just need to select a 'hardware' type that suits how the matrix is wired by rows/columns.

There should be lots on the web if you search or ask in the General Electronics section of the Arduino forum (forum.arduino.cc).

ieee488, 2019-04-14:
I am not sure what you mean select a 'hardware' type.
The hardware I have is FC-16 boards that I bought on eBay, but the seller sent me 1088B instead of 1088A LEDs.
I select FC16_HW in the library, but it doesn't work.

MajicDesigns, 2019-04-14:
You cannot just plug in 1088B matrix where a 1088A is expected. This is a different part and will not work (the diodes are the other way round). This is not a software problem - it is incompatible hardware. If you want to use the 1088B with a MAX7219 you have to design a completely different circuit using the MAX7219. It easier to just buy 1088A matrices instead and complain to the seller of the original hardware.

Please look at the data sheets for the matrices to understand what is happening.

ieee488, 2019-04-14:
That's what I thought. 
You may want to correct Paul__B's post #12 
https://forum.arduino.cc/index.php?topic=609299.0
where he says all one has to do is change something in the library.

.

closed 2019-04-25 by MajicDesigns