MajicDesigns/MD_MAX72XX #18: Missing semicolon at end of font structure written by txt2font.c (print statement at line 349)

nfhamilton, 2019-05-09:
## IMPORTANT
Before submitting this issue
[yes ] Have you tried using the latest version of the library?
[ yes] Have you checked this has not already been submitted and/or resolved?
[ ] If you are requesting help a better choice may be the [ Arduino forum](http://forum.arduino.cc/)

## Subject of the issue
The text in print statement in line 349 of "Font Builder/src/txt2font.c" is missing the trailing semicolon in the output font array definition. Needs semicolon after closing curly bracket.

Org line:
  fprintf(G.fpOut, "}\n\n");
Should be:
  fprintf(G.fpOut, "};\n\n");

## Your Environment
**Library Version: 3.0.2
**Arduino IDE version: 1.8.8
**Hardware model/type: WeMos D1 R1
**OS and Version: Ubuntu 18.04

## Steps to Reproduce
Build any font.h using the supplied example font.txt files. Include resultant font header file in user src. Compiling user src will issue what appears to be unrelated errors -- traceable back to the missing semicolon.

## Expected Behaviour
Array definition need to be terminated by a semicolon.

MajicDesigns, 2019-05-09:
Fixed. NIce to see someone is using these utilities!

closed 2019-05-09 by MajicDesigns

nfhamilton, 2019-05-09:
Glad I could help track it down.

Nice work on the library, BTW! Thanks!!!

https://github.com/MajicDesigns/MD_MAX72XX/issues/18#issuecomment-491041424

leosdad, 2023-08-25:
It is possible that the semicolon was added after txt2font.executable was generated, because the output is still missing the semicolon at the end:

```cpp
// Simplified for clarity
const uint8_t PROGMEM _font8_var_slots[] =  {  'F', 1, 32, 123, 8,  /* ... */ }
```

instad of

```cpp
// Simplified for clarity
const uint8_t PROGMEM _font8_var_slots[] =  {  'F', 1, 32, 123, 8,  /* ... */ };
```

I have a couple of simple suggestions too, please let me know if I should post them here.

Other than that, thanks a lot for your great work.

MajicDesigns, 2023-08-26:
Current code on github has the semicolon after the curly brace (line 349 of txt2font.c).

Ideas are welcome. Please put them into other issues, one per entry so they can be tracked properly.

leosdad, 2023-08-26:
> Current code on github has the semicolon after the curly brace (line 349 of txt2font.c).

Sure, but [the executable file](https://github.com/MajicDesigns/MD_MAX72XX/blob/main/Font%20Builder/txt2font.executable) does not. I believe it must be simply recompiled and updated.

> Ideas are welcome. Please put them into other issues, one per entry so they can be tracked properly.

I will, thanks.

MajicDesigns, 2023-08-27:
Yes, I understand now. Executable will be updated the next library update, whenever that happens.