MajicDesigns/MD_MAX72XX #21: ESP32 with Robot Eyes Example

Gabones, 2019-06-09:
## IMPORTANT
Before submitting this issue
[X] Have you tried using the latest version of the library?
[X] Have you checked this has not already been submitted and/or resolved?
[maybe] If you are requesting help a better choice may be the [ Arduino forum](http://forum.arduino.cc/)

## Subject of the issue
Robot Eye example is not compatible with ESP32, when i try to compile the code for ESP32 both in the Arduino IDE and in PlatformIO i get a lot of errors. If i try with arduino board the code compile normaly

## Your Environment
**3.1.0:** 
**Arduino IDE version: 1.8.21 - PlatformIO Home 2.1.0 · Core 3.6.7** 
**Hardware model/type: ESP32** 
**OS and Version:** 

## Steps to Reproduce
Load example, select Arduino Board, it compiles ok, select esp32 board, see the errors

## Expected Behaviour
i think de c++ code should be full functional to all microcontrollers , the other examples works fine

MajicDesigns, 2019-06-11:
Trivial changes due to unnecessary stricter const handling in all ESP compilers. Fixed. Will be included in next MD_MAX72xx release.

closed 2019-06-11 by MajicDesigns

Gabones, 2019-06-30:
Could I use MD_MAX72xx and the MD_Parola library to emulate the eyes and mouth of a robot through 4 devices connected in series? (my esp32 is full). I want the first two to be running the animations of the eyes and the last two the animation of the mouth. Could you give me a hint how to do it? Sorry for bothering you, but your previous tips were a great help and I'm a beginner.

MajicDesigns, 2019-06-30:
Can you please ask this question on the Arduino forum thread for MD_MAX72xx support. The answer is probably useful for may others as well. Link [https://forum.arduino.cc/index.php?topic=171056.1530](url)