MajicDesigns/MD_MAX72XX #22: How to revert to old font.

tavdog, 2019-07-19:
After updating I see the default font has changed.  How do I revert to the old font with the updated version ? Thanks.

MajicDesigns, 2019-07-20:
In MD_MAX72xx_font.cpp change "#define USE_NEW_FONT 1" to value of 0.

closed 2019-07-20 by MajicDesigns

tavdog, 2019-07-20:
Thank you so much.  Here a few images of my project using the MAX7219.  
[image]
[image]
[image]
[image]