MajicDesigns/MD_MAX72XX #24: MAX7219, 1088AS 4 x 8x8 mirrored text

knopserl, 2019-08-30:
I have a 4 devices 8x8 MAX7219 display block linke below:
https://de.aliexpress.com/item/32955063973.html?spm=a2g0s.9042311.0.0.27424c4dEH9hyR
I'm using an ESP8266, NodeMCU (LOLIN) and the newest 3.1 library of MD_MAX72XX and MD_Parola).
I'm using the Text-Scroll example and it works but the scrolling text is horizontally mirrored.
So each char is flipped 180degree horizontally. so e,r are shown to the left side instead of right side.

Any ideas?
I have tried the different HW models, but it get worse, if I use nay other model:
//    PAROLA_HW,    ///< Use the Parola style hardware modules.
//    GENERIC_HW,   ///< Use 'generic' style hardware modules commonly available.
//    ICSTATION_HW, ///< Use ICStation style hardware module.
//    FC16_HW       ///< Use FC-16 style hardware module.

knopserl, 2019-08-30:
Ok, I have tested again and with this HW settings and turing the display around it works:
#define HARDWARE_TYPE MD_MAX72XX::ICSTATION_HW

closed 2019-08-30 by knopserl

877dev, 2021-12-25:
I needed to use `FC16_HW` for my 8x32 Amazon display, thanks to this issue I realised the fix - thanks!