MajicDesigns/MD_MAX72XX #25: Under docs you need to rename the files so they do not start with an "_" because the Arduino IDE won't read the files.

MrWinnerson, 2019-10-31:
## IMPORTANT
Before submitting this issue
[ ] Have you tried using the latest version of the library?
[ ] Have you checked this has not already been submitted and/or resolved?
[ ] If you are requesting help a better choice may be the [ Arduino forum](http://forum.arduino.cc/)

## Subject of the issue
Describe your issue here.

## Your Environment
**Library Version:** 
**Arduino IDE version:** 
**Hardware model/type:** 
**OS and Version:** 

## Steps to Reproduce
Explain how to reproduce this issue. Please provide working code below to demonstrate the issue.

## Expected Behaviour
Explain what should happen.

## Actual Behaviour
Explain what happens instead. Provide log messages if relevant.

## Code Demonstrating the Issue
````
Insert your code here.
````

MajicDesigns, 2019-10-31:
What files are you trying to read from the docs folder using the IDE? These are html files for the library documentation and are not meant for the IDE.

Can you explain why this is a problem rather than give me a solution that I don't understand.

MrWinnerson, 2019-10-31:
I am sorry I did not mean to send that, my bad. I meant when I try to install your library on the IDE it gives me this warning:

The library "MD_MAX72XX 2, 3, 4, 5, and 6 cannot be used.
Library Folder names must start with a letter or number, followed by letters, numbers, dashes, dots and underscores. Maximum length is 63 characters.
[image]

MajicDesigns, 2019-10-31:
There seems to be 2 things being confused here. The pop-up message references the MD_MAXPanel library and your message talks about the MD_MAX72xx library. These are different.

However, in neither case are any of these followed by a space and/or commas and/or numbers. They are simply known by their name (MD_MAX72xx, MD_MAXPanel), so I really don't understand what the IDE is trying to tell you.

Cesar-S, 2020-01-21:
I have not tried it but there I see that it says "MD_MAXPANEL 3" there is a space, will that be why error?

MajicDesigns, 2020-01-21:
If your folder is called that then you have a installation issue. Have you really just installed the library using the IDE? Or are there other copies manually created?

Maybe you should delete all installations with any connection to MD_ and start again.

Cesar-S, 2020-01-21:
No, I don't have the problem ... I just commented.
 I think that is what happens when you try to load it into the IDE.

closed 2020-01-21 by mr-winson