MajicDesigns/MD_MAX72XX #31: Custom Sprites

ChopsKingsland, 2020-08-20:
Does anyone know how to display custom sprites on the MAX7219 using the MD_MAX72xx and MD_PAROLA libraries?

MajicDesigns, 2020-08-20:
None of the MD_Parola examples using sprites help?

ChopsKingsland, 2020-08-20:
No, because I don't really understand how they work...

ChopsKingsland, 2020-08-20:
[image]

I did find this program that someone wrote, but that uses the LedControl library

MajicDesigns, 2020-08-20:
Might be useful to read the article “Parola A to Z – Sprite Text Effects” on my blog site https:\\arduinoplusplus.wordpress.com

ChopsKingsland, 2020-08-20:
Is there anyway to do it so the sprite just stays on the matrix, and isn't a text effect?

MajicDesigns, 2020-08-21:
Load it column by column using MD_MAX72XX. With Parola it assumes you will do animations.

ChopsKingsland, 2020-08-21:
How do I load my 'pattern' column by column?

MajicDesigns, 2020-11-22:
Close due to inactivity

closed 2020-11-22 by MajicDesigns