MajicDesigns/MD_MAX72XX #32: intensity 0 is still too 'intense' ;-)

gertst, 2020-08-28:
Hi Marco,

If I use matrix.control(MD_MAX72XX::INTENSITY, 0); to set the intensity to the lowest value, it is still to bright when used for a clock in a bedroom. 

How can I 'hack' this to lower the brighness? Since the required intensity value is an uint I can't use negative values or floats, but may be there is an easy way to hack it, before I'm diving in your source code ;-)

I'm using this display: [DAOKI MAX7219 Dot led Matrix MCU 8x32 Control LED Display Module Drive for Arduino Raspberry Pi 4 in 1 (Brand: DAOKI)](https://www.amazon.com/gp/product/B07X95FHXV/ref=ppx_yo_dt_b_asin_title_o01_s00?ie=UTF8&psc=1)

Thanks a lot!
Gert

MajicDesigns, 2020-08-28:
Use shutdown() to turn off the display.

gertst, 2020-08-28:
I saw the shutdown method, but in fact I hope to find a way to keep the display on, but with an intensity even lower then 0 ...

MajicDesigns, 2020-08-28:
Intensity is governed by the resistor on the module board. The 10 k typical from commercial suppliers is way too low and makes the display too bright. You need to change the resistor. See the data sheet for the details, look for Rset. It has a table of values. On my displays I use 69 k.

MajicDesigns, 2020-11-22:
Closed stale thread

closed 2020-11-22 by MajicDesigns