MajicDesigns/MD_MAX72XX #38: Library doesn't compile for Arduino Nano 33 BLE

tim-vandecasteele, 2020-12-06:
## IMPORTANT
Before submitting this issue
[x] Have you tried using the latest version of the library?
[x] Have you checked this has not already been submitted and/or resolved?
[x] If you are requesting help a better choice may be the [ Arduino forum](http://forum.arduino.cc/)

## Subject of the issue
Trying to compile `MD_MAX72xx` for Nano 33 BLE gives errors that SPI is not defined. This is because #30 introduces compatibility with mbed. The Arduino Nano 33 BLE also runs mbed, and this causes the`__MBED__` macro to be defined. 

A possible solution to this would be to check for `#ifndef Arduino_h` and only check for `elseif __MBED__`. Although I'm not experienced with arduino and I'm not sure this would be bad practice.

When removing the `__MBED__` directives, I succeed to compile, but I haven't deployed it yet on a target.

I'm happy to work on a solution, but wanted to check what you think would be a good way forward?

## Your Environment
**Library Version:** 3.2.4
**Arduino IDE version:** 1.8.13 
**Hardware model/type:** Arduino Nano 33 BLE
**OS and Version:** Mac OS Big Sur

## Steps to Reproduce
Compile the below block for Arduino Nano 33 BLE

## Expected Behaviour
Should just compile.

## Actual Behaviour
Gives following error

```
Arduino: 1.8.13 (Mac OS X), Board: "Arduino Nano 33 BLE"

In file included from /var/folders/b0/57dvvvtn5l93d5m84k7vfrzh0000gn/T/arduino_modified_sketch_871210/BareMinimum.ino:1:0:
/Users/tim/Documents/Arduino/libraries/MD_MAX72XX/src/MD_MAX72xx.h:998:3: error: 'SPI' does not name a type; did you mean 'SPIS'?
   SPI   _spi;           // Mbed SPI object
   ^~~
   SPIS
/Users/tim/Documents/Arduino/libraries/MD_MAX72XX/src/MD_MAX72xx.h:999:3: error: 'DigitalOut' does not name a type; did you mean 'digitalRead'?
   DigitalOut _cs;
   ^~~~~~~~~~
   digitalRead
exit status 1
Error compiling for board Arduino Nano 33 BLE.
```

## Code Demonstrating the Issue
````
#include <MD_MAX72xx.h>
#include <SPI.h>

void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}
````

MajicDesigns, 2020-12-06:
So to be clear the error is because you don't want to run embed OS but 'normal' Arduino compile?

In this case, how come the __EMBED__ is defined if embed is not used? Usually it is defined in a header file for the product or on the command line for the compiler.

JojoS62, 2020-12-06:
The Arduino Nano 33 BLE is using Mbed internaly, so this macro is defined. I will look for a solution.

JojoS62, 2020-12-06:
I've provided a fix for this. It compiles, but I don't have the hardware to test this. 
@tim-vandecasteele could you thest this on your Nano 33 BLE?
And thanks for your detailed issue decription.

MajicDesigns, 2020-12-06:
Released as version 3.2.5

closed 2020-12-19 by MajicDesigns