MajicDesigns/MD_MAX72XX #44: ESP8266 and MAX72xx

redguyz, 2021-09-09:
## IMPORTANT
Before submitting this issue
[X ] Have you tried using the latest version of the library?
[X ] Have you checked this has not already been submitted and/or resolved?
[ ] If you are requesting help a better choice may be the [ Arduino forum](http://forum.arduino.cc/)

## Subject of the issue
AP of the ESP is still active after connection to wireless network.
## Your Environment
**Library Version:** 
**Arduino IDE version:** 
**Host OS and Version:** 
**CPU Hardware model/type:** 
**Matrix hardware model/type:** [ ] Parola  [ ] Generic  [ ] ICStation  [ ] FC-16  [ ] Custom (describe how wired)

## Steps to Reproduce
Simply flash the example code.

## Expected Behaviour
Connection to wireless network should deactivate AP mode of ESP.

## Actual Behaviour
Connection to wireless network is done. AP mode of ESP is active also.

## Code Demonstrating the Issue
````
Insert your compilable code here (no code snippets).
````

redguyz, 2021-09-09:
Thank you for a great piece of code. Is it possible the AP mode of ESP8266 to be disabled after connection to wireless network?
Regards,

Ivaylo

MajicDesigns, 2021-09-10:
What bit of code are you specifically referring to?

redguyz, 2021-09-10:
Excuse me, I mean "MD_MAX72XX_Message_ESP8266.ino"

MajicDesigns, 2021-09-10:
OK. If you know how this is done then you can submit the code and I will review it and update the example. If you are asking me to do this it will go on my 'to maybe do' list and I can definitely say that it will be a very low priority, as this is just example code to show the basics and for others to start from.

closed 2021-09-16 by redguyz