MajicDesigns/MD_MAX72XX #45: MD_MAX72XX_Message_ESP8266 script

redguyz, 2021-09-17:
## IMPORTANT
Before submitting this issue
[X] Have you tried using the latest version of the library?
[X] Have you checked this has not already been submitted and/or resolved?
[ ] If you are requesting help a better choice may be the [ Arduino forum](http://forum.arduino.cc/)

## Subject of the issue
Describe your issue here.

## Your Environment
**Library Version:** 
**Arduino IDE version:** 
**Host OS and Version:** 
**CPU Hardware model/type:** 
**Matrix hardware model/type:** [ ] Parola  [ ] Generic  [ ] **ICStation**  [ ] FC-16  [ ] Custom (describe how wired)

## Steps to Reproduce
Enter in browser any symbol beyond 159 (9F) and it is not displayed correctly. It shows the code e.g. %A0 but not the actual symbol.

## Expected Behaviour
Show all symbols from 00 to FF.

## Actual Behaviour
Only symbols from 00 to 9F are displayed. Others, from A0 to FF are not shown (%A0 to %FF on MAX7219).

## Code Demonstrating the Issue
````
Insert your compilable code here (no code snippets).
````

redguyz, 2021-09-17:
Is it possible to fix?

MajicDesigns, 2021-09-18:
It looks like the HTML transaction is replacing the characters with a representation of the characters.

When you receive the message from the web page into the microcontroller, you need to parse the string and replace all %xx in the string with the character whose code is xx (a hex number). You should investigate what happens when you type a % as this may be sent as %% and will also need to be treated/converted.

Note that the result string will always be less than or equal to the length of the received string, so you can so the replacement within the receive buffer and not allocate a new buffer.

MajicDesigns, 2021-09-29:
Closed as no further response from OP

closed 2021-09-29 by MajicDesigns