MajicDesigns/MD_MAX72XX #46: Name Of the Default Font

sheminasalam, 2021-09-22:
What is the name of the default font in this library? Where can I find the ttf file for that?

MajicDesigns, 2021-09-22:
There is no TTF file. All the data for the font is found in the font.cpp source file. Eternal tools for building the font in this format are described in the documentation and on this blog post [https://arduinoplusplus.wordpress.com/2016/11/08/parola-fonts-a-to-z-defining-fonts/](url)

closed 2021-09-22 by MajicDesigns

sheminasalam, 2021-09-23:
I have made a ttf font file out of the data from the font file in this library which has been useful in compiling with some firmwares that depend on this library such as esphome. Anyone who want to try this can download the ttf file I have shared here. Also please make the font size as 8 if you are using 8x8 matrix. A heads up, some symbols in the font are not recognized by esphome.
[matrix_led.zip](https://github.com/MajicDesigns/MD_MAX72XX/files/7214528/matrix_led.zip)