MajicDesigns/MD_MAX72XX #48: Allow font data to be stored in RAM (ideally loaded from a Stream)

bblanchon, 2022-02-26:
Hi,

Thank you very much for this excellent library.

My client wants to customize the font without recompiling the project.
I searched for a way to load the font data from a binary file on an SD card.
I looked at `MD_MAX72xx_font.cpp` and saw that the library assumes that the font is in program memory.

Unfortunately, this design prevents customizing the font at runtime.
Could you provide a way to support this scenario?

I wish we could directly pass the `File` instance to `setFont()`, but passing a pointer would be a good first step.

Best regards,
Benoit

MajicDesigns, 2022-02-27:
As designed the font must be in program memory. You could add additional functions that would handle RAM based font (data table) or just change the while library to use a ram based font instead. Characters are loaded one character at a time so once the character is loaded, everything else is already referring to the copy in RAM. See this blog post for an explanation [](https://arduinoplusplus.wordpress.com/2016/11/13/parola-fonts-a-to-z-managing-fonts/)

bblanchon, 2022-02-27:
Are you saying that I must fork the library to support this feature?

MajicDesigns, 2022-02-27:
Yes, or just create your own version. At this stage I don't have the time to do anything about it.

closed 2022-04-07 by MajicDesigns

bblanchon, 2022-04-07:
Correct me if I'm wrong, but the LGPL license forces me to release any modified version of the library under the LGPL license.
Therefore, I cannot just "create me own version"; I'm also obliged to make it open-source.

MajicDesigns, 2022-04-07:
You are right. However, if you are not distributing source code, you are free to include the modified compiled code in your works.

And you can also modify and do a pull request for it to be included into the main library. That is the point of open source.

bblanchon, 2022-04-08:
It wasn't clear to me that you would accept a pull request on something that could have such a significant impact on the code.
After all, open-source doesn't mean open contributions.
But that's good to know. 

Fortunately for me, my client has accepted the current solution with a predefined font; I'll get back to you if this feature comes back.