MajicDesigns/MD_MAX72XX #49: Library does not control large displays (eg 28 x FC16_HW)

RalphBacon, 2022-09-11:
## Subject of issue
Sketch/library **MD_MAX72xx_Test.ino** works perfectly (with ESP-specific SPI pins) on **8 x FC-16** devices. When used on my 1 metre long MAX7219 strip (**28 devices**) it fails to control the last 6-8 modules (gets progressively worse, the longer the sketch runs for). Powered via a separate 3v3 supply in case this was causing an issue (and the ESP32's onboard 3v3 regulator cannot support this current requirement anyway).

## Set up details
Using MD_MAX72XX library version 3.3.0 (taken from library.json).

Running an unchanged example sketch **MD_MAX72xx_Test.ino** in a Visual Studio Code (PlatformIO) environment with the Arduino framework on an ESP32 Dev Module. Windows 11 Home. FC-16 devices.

## Date of issue
11th September 2022.

## Notes & evidence
I have tested the 28 device strip using an Arduino UNO and a very simple sketch (not using this library) and it correctly displays/scrolls the entire 28-device length so I believe the hardware is wired up & working correctly.

What is happening: The first 20 approx device (counting from the right) are correctly controlled with text, graphics etc. The remaining device remain either full lit (all LEDs on or off, varies). The graphics seems to run into the devices that are permanently lit. 

**Video demo** (speeded up) here: https://www.dropbox.com/s/eidwwuvr9cvlp7k/MAX7219.mp4?dl=0

Best Regards
Ralph S Bacon

MajicDesigns, 2022-09-11:
The MAX7219 IC requires 5V power supply to work correctly. If you are powering it with 3.3V is not to the specification for this hardware. You may (and many do) get away with 3.3V for a few devices but signal degradation and voltage drops will quickly catch up with this direct connect method.

You will need to supply 5V to the MAX7219 and use level converters for the signal voltages (ie, the correct way to to do this).

RalphBacon, 2022-09-12:
## Results
Thanks, Marco. Ironically, my full circuit already included a level shifter and 5v so I have implemented these in the prototype too, as per your suggestion.

Also, each bank of 4 FW-16 devices has a VCC connection to an additional common 5v supply rail, as the total current consumption peaks at around 1.9A in my 28-device setup. A return ground has also been connected but I believe this would be better off as a beefed up connection _between_ individual FW-16 devices, as the track width is very small. I shall do this in my project.

Assuming I am successful in my project I shall certainly reinforce this message to my viewers.

Anyway, running on 5v with a level shifter made a big difference and the tests _almost_ run correctly.

## Failed tests
The tests that do not run/complete are in the "Control" set. The "intensity" seems to work, but the "scanLimit" blanks out half way through and then "corrupts" the entire display by only allowing a single FW-16 device to work thereafter; "blinking" works until the blinking gets to about 2fps then it just stops and blanks out the display for a few seconds.

I'm not too worried about the failed tests. My needs are modest and unless I come across the problem in my final project we can just ignore those features.

If you want me to video this just let me know.

## Final suggestion
Change the comment in the (exmple) files that says:
```
"LED matrices seem to work at 3.3V"
```
to something that says 
```
"A few LED matrices seem to work at 3.3V but longer runs 
will require 5v, better wiring and a PSU capable of supplying
the necessary current, to prevent the voltage dropping"
```

## Thank you
These libraries are very flexible, powerful, well-documented and, of course, well written 😉

MajicDesigns, 2022-09-12:
I suspect that if you are going in/out/in/out between the modules for the control SPI signals then there is degradation between the start and end module. All signals (except the DATA OUT/IN) can be connected in parallel, like you have done for the power line (maybe every 8 modules). For long runs this would be a good idea. The DATA lines, clearly, only go from one module to the next as they are generated by the IC.

RalphBacon, 2022-09-14:
I'll do this to prove it one way or the other! 

Not that I am doubting your for a minute, as it seems very plausible. But quite annoying that this does not seem to be documented by the seller of these devices as a caveat for longer runs. 

OTOH I don't suppose many people have runs of 28 runs or more. 🧐

If we can keep this issue open until I have done some testing that would be great.

MajicDesigns, 2022-09-14:
Sellers don't care - this is basic electrical engineering.

closed 2022-09-21 by MajicDesigns