MajicDesigns/MD_MAX72XX #51: Multiple Row Displays

TobiasPSP, 2023-01-10:
First, this is an awesome work,. Have reviewed multiple libs so far, yet this seems to me like the most professional and well-thought concept.

What I cannot seem to find is a way to use multiple rows, i.e. a 8x64 strip and another 8x64 strip to create one display of 16x64. I did look at zones, and also at addressing different strips separately. Yet what I am trying to do is display one text just in a bigger font. 

Is there a way, and could you point me to the appropriate example? I'd greatly appreciate it.

MajicDesigns, 2023-01-10:
Zones is a MD_Parola feature, so I assume you mean that you tried it in Parola, not his library.

You don't mention if you have tried the double height Parola examples (MD_Parola_Double_Height_*). This gives you 16 LED character height. It has some limitations but it seems to work ok for many users.

Just using the MD_MAX72xx library there is no simple way to do this and you will need to code it all.

If you want to create a single display for all the LEDs, the MD_MAXPanel library is most appropriate. However, the fonts are still handled the same way as for MD_MAX72xx and MD_Parola. It is far better for graphics applications (eg, simple games).

TobiasPSP, 2023-01-13:
Thank you so much. I did try MD_Parola_Double_Height, and at first it works beautifully.
However, once I started to make adjustments and switched the scroll mode to a static display (plus setting PAUSE_TIME from 0 to 3000)

```
P.displayZoneText(ZONE_UPPER, msgH, PA_CENTER, 0, PAUSE_TIME, PA_PRINT, PA_NO_EFFECT);
P.displayZoneText(ZONE_LOWER, msgL[cycle], PA_CENTER, 0, PAUSE_TIME, PA_PRINT, PA_NO_EFFECT);
```

I run into a strange problem: once I disconnect Arduino from power and then reconnect, the display is scrambled and won't run again (even if I re-upload the sketch with my adjustments). 

I need to upload the original scatch with the scrolling text to get back a working display.

I chewed on the code for some hours now but can't seem to find the reason why changing the scroll mode would interfere with correct device initialization after power loss. 

Would you have guidance for me? Many thanks!

MajicDesigns, 2023-01-14:
Make sure the power supply is working correctly. If you are using a 3V processor you will need a separate power supply for the display (5V) and a signal converter to provide the right signal voltage for the MAX chips. See [this blog post](https://arduinoplusplus.wordpress.com/2020/08/29/parola-a-to-z-frequently-asked-issues/) for more information.

TobiasPSP, 2023-01-14:
Thank you very much, you provided a very helpful link. 

I did suspect power issues before and had added an external power supply to my 5v project. I connected it to Vin though when in reality I should probably have also connected it directly to the display.

The (working) original sketch with scrolling text slowly lights an increasing # of LEDs whereas my static text example immediately shows the entire text - drawing much more power during startup.

it seems that the system becomes unstable when the displays sucks much energy/lights many segments *during startup only*. 

Later on, all segments can light up with no issue at all.

Thank you so much for pointing me to the right direction.

closed 2023-01-14 by TobiasPSP