MajicDesigns/MD_MAX72XX #52: MD_Parola initialization fails after reboot

TobiasPSP, 2023-01-14:
I randomly get issues with any example code that uses zones: most of the time, sketches work fine after uploading (every now and then even a fresh upload fails, though, and keeps failing until I upload a different example or manually change the number of zones or modules to a smaller number). 

When rebooting Arduino by powering it on, issues start to occur on a regular basis. Sometimes things work, yet most of the time a previously running sketch will now just distort the display. When that happens, the only way to recover is to upload a different sketch that uses less modules.

I already tried all suggestions given: added a strong external power supply, added direct power cables to the display, started the display blank in an effort to reduce power requirements initially. No improvement.

I then googled and found other users experiencing the very same issue. So apparently, when using more than just 1 or two modules or when starting to use zones, there seems to be some initialization issue, and initialization seems to be highly sensitive, I just can't seem to figure out what exactly is interfering.

I next wanted to try to init the library after a short delay, i.e. adding time for the power supply and SPI to stabilize.
Adding a delay() to setup() doesn't change anything, though, which is not surprising since the library is implicitly initialized at startup:

```
MD_Parola P = MD_Parola(HARDWARE_TYPE, CS_PIN, MAX_DEVICES);
```

Is there a way to postpone the library initialization somehow? I am no c++ guru so my experiments to move the above call to other places in my code, starting with an empty global var and filling it later, failed miserably.

I really love the library. If the hardware won't start reliably from a power-off situation, though, my project would be at a sudden end. Do you have any other hint or suggestion for me?

Many thanks.

MajicDesigns, 2023-01-14:
Can you please raise this issue in the Parola library repository. Also please indicate what hardware and connections you have.

TobiasPSP, 2023-01-14:
Many thanks for your quick response. Do you have a link to the place where you want me to post this issue?

I am using an Arduino Mega and a 2x10 8x8 matrix. I can now reliably repro the case:

1. Sketch (code below) runs fine. I can press the Arduino reset button, and it still runs fine.
2. Once I turn off power and power on again, the display stays blank or is distorted most of the tries. Once the LEDs do become unresponsive, there is no way for me to recover (by pressing reset or uploading the sketch again).
3. To recover and get the LEDs responding again, in the sketch I need to change ZONE_SIZE to a *different* number (lower or higher makes no difference as long as it differs from the currently used value). When I then upload the temporarily adjusted sketch, the LEDs respond properly again. After I get them working again, I can then revert the ZONE_SIZE back to its original value, and my panel is working perfectly again - until the next time I power off.

Here is my code which is a trivial adaption of Parola_Double_Height_V2:

```
#include <MD_Parola.h>
#include <MD_MAX72xx.h>
#include <SPI.h>
#include "Font_Data.h"

// Define the number of devices we have in the chain and the hardware interface
// NOTE: These pin numbers may not work with your hardware and may need changing
#define HARDWARE_TYPE MD_MAX72XX::FC16_HW
#define NUM_ZONES 2
#define ZONE_SIZE 10
#define MAX_DEVICES (NUM_ZONES * ZONE_SIZE)

#define ZONE_UPPER  1
#define ZONE_LOWER  0

#define PAUSE_TIME 2000

#define SCROLL_SPEED 50

#define CLK_PIN   50
#define DATA_PIN  51
#define CS_PIN    49

textEffect_t scrollUpper = PA_SCROLL_LEFT;
textEffect_t scrollLower = PA_SCROLL_LEFT;

// HARDWARE SPI
MD_Parola P = MD_Parola(HARDWARE_TYPE, CS_PIN, MAX_DEVICES);
bool isReverse = false;
bool scroll = false;

// SOFTWARE SPI
//MD_Parola P = MD_Parola(HARDWARE_TYPE, DATA_PIN, CLK_PIN, CS_PIN, MAX_DEVICES);

const char *msgL[] =
{
  " ",
  "Emergency",
  "EMERGENCY",
};
char *msgH; // allocated memory in setup()

void setup(void)
{
  uint8_t max = 0;

  // work out the size of buffer required
  for (uint8_t i = 0; i<ARRAY_SIZE(msgL); i++)
    if (strlen(msgL[i]) > max) max = strlen(msgL[i]);

  msgH = (char *)malloc(sizeof(char)*(max + 2));

  // initialise the LED display
  P.begin(NUM_ZONES);

  // Set up zones for 2 halves of the display
  P.setZone(ZONE_LOWER, 0, ZONE_SIZE - 1);
  P.setZone(ZONE_UPPER, ZONE_SIZE, MAX_DEVICES-1);
  P.setFont(BigFont);
  P.setCharSpacing(P.getCharSpacing() * 2.0); // double height --> double spacing
  P.setIntensity(15);
 
  isReverse = true;
}

void createHString(char *pH, const char *pL)
{
  for (; *pL != '\0'; pL++)
    *pH++ = *pL | 0x80;   // offset character

  *pH = '\0'; // terminate the string
}

void loop(void)
{
  static uint8_t cycle = 0;

  P.displayAnimate();

  if (P.getZoneStatus(ZONE_LOWER) && P.getZoneStatus(ZONE_UPPER))
  {
    // set up the string
    createHString(msgH, msgL[cycle]);

    P.displayClear();
    
    if (scroll)
    {
      P.displayZoneText(ZONE_LOWER, msgL[cycle], PA_LEFT, SCROLL_SPEED, 0, scrollLower, scrollLower);
      P.displayZoneText(ZONE_UPPER, msgH, PA_LEFT, SCROLL_SPEED, 0, scrollUpper, scrollUpper);
    }
    else
    {
      P.displayZoneText(ZONE_LOWER, msgL[cycle], PA_CENTER, 0, PAUSE_TIME, PA_PRINT, PA_NO_EFFECT);
      P.displayZoneText(ZONE_UPPER, msgH, PA_CENTER, 0, PAUSE_TIME, PA_PRINT, PA_NO_EFFECT);
    }
     
// synchronize the start and run the display
    P.synchZoneStart();

    // prepare for next pass
    if (isReverse)
    {
      cycle = (cycle + 1) % ARRAY_SIZE(msgL);
      if (cycle == 0) { cycle = 1; }
      isReverse = false;
      P.setZoneEffect(ZONE_UPPER, true, PA_FLIP_LR);
      P.setZoneEffect(ZONE_LOWER, true, PA_FLIP_LR);
    }
    else
    {
      isReverse = true;
      P.setZoneEffect(ZONE_UPPER, false, PA_FLIP_LR);
      P.setZoneEffect(ZONE_LOWER, false, PA_FLIP_LR);
    }   
  }
}
```

In a nutshell:
- Device initialization during Arduino power-on seems very fragile and can result in a non-responsive hardware state. Hardware stays in this responsive state.
- Device initialization in a *running* Arduino (i.e. by pressing Reset, uploading new sketches) always works well.
- Once the issue has hit during a power on, uploading the SAME sketch again will not repair the issue. The hardware will stay unresponsive. Only when the number of declared devices is changed in the sketch (higher or lower) and this adjusted sketch is uploaded in the stable condition of an already running Arduino will the hardware be repaired and becomes responsive again.

So I really would love to try and initialize the hardware after a small delay, as this would verify the findings above and also provide a workaround. 

Do you have an idea how I could achieve this delay in my code?

MajicDesigns, 2023-01-15:
You are posting in the issues section for the MD_MAX72xx library (https://github.com/MajicDesigns/MD_Max72XX). The library you are discussing is the MD_Parola library (https://github.com/MajicDesigns/MD_Parola). Please post this in the correct repository so that it is on-topic and others can find this issue in future.