MajicDesigns/MD_MAX72XX #54: Corruption in scrolled text example

elan, 2023-06-01:
Scrolling text can leave garbage in-between characters. This is due to the code here:

```c++
    for (uint8_t i=0; i<=charWidth; i++)
    {
      mx.transform(MD_MAX72XX::TSL);
      if (i < charWidth)
        mx.setColumn(0, cBuf[i]);
      delay(DELAYTIME);
    }
```

The correct code should be like this (we need to zero out the row explicitly):

```c++
    for (uint8_t i=0; i<=charWidth; i++)
    {
      m_mx->transform(MD_MAX72XX::TSL);
      m_mx->setColumn(0, (i < charWidth) ? cBuf[i] : 0);
      delay(DELAYTIME);
    }
```

Thanks for a great library!

MajicDesigns, 2023-06-01:
Which example file? (name)

elan, 2023-06-01:
I apologize for the lack of context: https://github.com/MajicDesigns/MD_MAX72XX/blob/a761e79a2cf29538cbbdb0c0cb0754ac7ddc8612/examples/MD_MAX72xx_Test/MD_MAX72xx_Test.ino#L53-L58

MajicDesigns, 2023-06-02:
Change will be included in the next release.

closed 2023-06-02 by MajicDesigns