MajicDesigns/MD_MAX72XX #55: Non-sequential rows/columns

skullydazed, 2023-07-30:
## IMPORTANT
Before submitting this issue
[x] Have you tried using the latest version of the library?
[x] Have you checked this has not already been submitted and/or resolved?
[x] If you are requesting help a better choice may be the [ Arduino forum](http://forum.arduino.cc/)

## Subject of the issue
I found a handful of surplus 8x8 LED modules in my stuff, and designed a MAX7219 board to drive them. I didn't pay much attention to the row/col ordering because in other LED projects I've used it's simple enough to change that mapping.

I can't seem to figure out how to change that mapping for MD_MAX72XX. I have this matrix that I want to drive with Parola:

```
      | Dig-3 | Dig-6 | Dig-2 | Dig-1 | Dig-7 | Dig-5 | Dig-4 | Dig-0
SEG-B |
SEG-F |
SEG-G |
SEG-D |
SEG-E |
SEG-C |
SEG-A |
SEG-DP |
```

How can I define that matrix in MD_MAX72XX?

## Your Environment
**Library Version:** Latest downloaded by Arduino today
**Arduino IDE version:** 2.0
**Host OS and Version:** KDE Plasma (Linux 5.19.0)
**CPU Hardware model/type:** Teensy 2.0++
**Matrix hardware model/type:** [ ] Parola  [ ] Generic  [ ] ICStation  [ ] FC-16  [x] Custom (describe how wired)

## Steps to Reproduce
Explain how to reproduce this issue. Please provide working code below to demonstrate the issue.

## Expected Behaviour
Explain what should happen.

## Actual Behaviour
Explain what happens instead. Provide log messages if relevant.

## Code Demonstrating the Issue

This is how I mapped out the rows/cols
````
#include <MD_MAX72xx.h>
#include <SPI.h>

#define HARDWARE_TYPE MD_MAX72XX::GENERIC_HW
#define MAX_DEVICES 1
#define CS_PIN 20
#define DATA_PIN 22
#define CLK_PIN 21
MD_MAX72XX myRawDisplay = MD_MAX72XX(HARDWARE_TYPE, CS_PIN, MAX_DEVICES);

void setup() {
  Serial.begin(57600);
  Serial.print("Starting test...\n");
  myRawDisplay.begin();
}

void loop() {
  for (int x=0; x < 8; x++) {
    for (int y=0; y < 8; y++) {
      Serial.print("Turning on row:");
      Serial.print(x, DEC);
      Serial.print(" col:");
      Serial.print(y, DEC);
      Serial.print("\n");
      myRawDisplay.setPoint(y, x, true);
      while(Serial.available() == 0) {
      }
      int mydata = Serial.read();
      myRawDisplay.setPoint(y, x, false);
    }
  }
}
````

MajicDesigns, 2023-07-30:
The MD_MAX72xx library does not cater for nonsequential mapping like you have created.

Changing it to be more general would be a ton of work. I think it would be easier to rewire your matrices or find another MAX72xx library that does support random mappings. I don't know any that do, so if you find one please let me know.

skullydazed, 2023-07-31:
Thanks Marco. I think you're right, in the Arduino world everyone seems to make the hardware align. I come from a custom keyboard background and in the keyboard firmware world it's common have two arrays, one for the actual data to be sent and another for the ordering of the LEDs. There's a small performance overhead in populating the second array but it's never been a problem even with complicated animations. Even an atmega32u4 was able to run complicated animations while also scanning a keyboard matrix.

I guess I will order new boards, luckily I only soldered one of them so desoldering it shouldn't be too bad.

Thanks for the answer!

closed 2023-07-31 by skullydazed