MajicDesigns/MD_MAX72XX #57: esp32

roetske, 2023-09-21:
esp32 dev mod 
i get when compiling this warning 
libraries\MD_MAX72XX\src/MD_MAX72xx.h:14:2: warning: #warning "ARDUINO SPI interface activated." [-Wcpp]
 #warning "ARDUINO SPI interface activated."
I notice also spi is runnning slow. Speed scrolling stays the same. even if i change speed to 25.
 
My includes are the following.
#include <MD_Parola.h>
#include <MD_MAX72xx.h>
#include <SPI.h>
#include <elapsedMillis.h>
#include <RTClib.h>
#include <WiFiManager.h>

MajicDesigns, 2023-09-21:
This is just an information message to verify that the ARDUINO option is enabled. I comes from this preprocessor code:
````
#if MBED_SPI_ACTIVE
#warning "MBED SPI interface activated."
#else
#warning "ARDUINO SPI interface activated."
#endif
````
If your SPI is running slow then you will need to check other sections of your code as this is probably not the cause.

roetske, 2023-09-24:
thx for the reply.

closed 2023-09-24 by MajicDesigns