MajicDesigns/MD_MAX72XX #58: The Excel Font Generator Tool "copy to clipboard" doesn't work on windows 10. But I found a solution.

Cherianpaul, 2023-11-17:
## IMPORTANT
Before submitting this issue
[*] Have you tried using the latest version of the library?
[*] Have you checked this has not already been submitted and/or resolved?
[ ] If you are requesting help a better choice may be the [ Arduino forum](http://forum.arduino.cc/)

## Subject of the issue
The Excel Font Generator Tool "copy to clipboard" doesn't work on windows 10. But I found a solution.

## Your Environment
**Library Version:** 
**Arduino IDE version:** 
**Host OS and Version:** 
**CPU Hardware model/type:** 
**Matrix hardware model/type:** [ ] Parola  [ ] Generic  [ ] ICStation  [*] FC-16  [ ] Custom (describe how wired)

## Steps to Reproduce
It is not with Arduino code, it is about the Excel Font Generator Tool.

## Expected Behaviour
It should copy the generated array into clipboard.

## Actual Behaviour
It does not copy as obj.PutInClipboard does not work on windows 10. 

## Code Demonstrating the Issue
````
MyDataObj.PutInClipboard
````

Cherianpaul, 2023-11-17:
Solution : https://chandoo.org/forum/threads/clipboard-copy-vba-code-not-working-in-windows-10.37126/post-223256

MajicDesigns, 2023-11-17:
This will need fixing. I don't have access to Microsoft Office any more so it may take a while. Are you able to implement a fix and submit?

Cherianpaul, 2023-11-17:
Sure. No problem. 
Can I attach it here? I am not an expert in Github usage.

MajicDesigns, 2023-11-17:
Better you email the excel file directly to me.

Cherianpaul, 2023-11-17:
Your email?

MajicDesigns, 2023-11-17:
Oops, forgot that I made it private here. OK, try and attach the file to one of these comments. The toolbar above the text as you type has a paperclip for attachment. Not sure if there is a size limit.