MajicDesigns/MD_MAX72XX #59: md_max72xx message esp8266.ino not working

mjmokhtar, 2023-11-29:
I just compile but my ip not show web respond and also web page

are you sure that's correct ?

please you try its not working trust me

MajicDesigns, 2023-11-30:
1. Did you put in your network credentials in the code:
`
// WiFi login parameters - network name and password
const char* ssid = "";
const char* password = "";
`

3. Have you tried turning on DEBUG to see if there are any information messages:
`
#define DEBUG 1
`

mjmokhtar, 2023-12-01:
I've using esp32 and tried using debug, but its IP doesn't work

MajicDesigns, 2023-12-01:
You need to be more specific. What do you mean by 'it does not work'. Does it get a wrong IP, no IP, what?

MajicDesigns, 2023-12-01:
What browser are you using to open the page? Have you tried a different one or a different computer?

mjmokhtar, 2023-12-01:
i've been try using safari and firefox

MajicDesigns, 2023-12-02:
I have checked the MD_MAX72xx_Message_ESP8266 and it works ok for me. While doing this I have also changed it to be MD_MAX72xx_Message_ESP32 as this is more common now, released in code tagged v3.5.1.

Tested with ESP32 Generic Development board. Connected with browsers below and all worked to change the displayed text on dot matrix:
- Microsoft Edge on Windows 11
- Google Chrome on iPhone
- Apple Safari on iPhone

closed 2023-12-24 by MajicDesigns