MajicDesigns/MD_MAX72XX #61: Sources for custom max7219 display boards ?

tavdog, 2024-01-10:
Just a quick question if you know how to get custom version of the popular 32x8 dot matrix boards powered with 4 max7219 chips.  They are so cheap on amazon but I want one with a built in esp32 or at least with straight headers instead of 90 degree headers. Any ideas ?

MajicDesigns, 2024-01-10:
I have not seen any matrix with an integrated cpu, mainly because they are designed to be ganged up end to end and you don’t need multiple processors for multiple matrix arrays.

Most of the LED matrices I buy have the headers supplied separately and you have to solder them in yourself, so you can make them whatever you need (ie substitute your own).

tavdog, 2024-01-11:
Can you give me the name of a supplier who ships the boards without presoldered headers ?

MajicDesigns, 2024-01-11:
Just look for max7219 on AliExpress. They show it separate if that is how they supply.

closed 2024-01-11 by tavdog