MajicDesigns/MD_MAX72XX #62: Flash on initialization for MAX7219

elan, 2024-01-12:
## Subject of the issue

With MAX7219, initialization can cause a quick flash in the display. The reason is this line here:

https://github.com/MajicDesigns/MD_MAX72XX/blob/main/src/MD_MAX72xx.cpp#L119

This causes the MAX7219 to latch in random data. Removing the line fixes the issue.

MajicDesigns, 2024-01-13:
Never seen this happen, but it is possible. What hardware are you using and are you implementing the slower software SPI interface?

elan, 2024-01-16:
Custom hardware, with hardware SPI.

MajicDesigns, 2024-01-21:
Checking in the data sheet, data is latched in to the MAX72xx hardware on the rising edge of the CS pin, so what is described is a possible scenario. Eliminating the line as proposed or setting the I/O to LOW are the solution. Code will be changed and published in the next release (low priority).

closed 2024-01-21 by MajicDesigns