MajicDesigns/MD_MAX72XX #65: CC2642R: Issue Integrating MAX72xx Library with CC2642R1F Microcontroller

Mahantesh-Navi, 2025-04-17:
Dear Team,
I’m working on integrating the MD_MAX72xx LED matrix driver library with the CC2642R1F. The integration is successful, and the code compiles without any issues. However, when I run the application and verify the code flow, I’m encountering a problem during execution.

Specifically, inside the setC method, the code flows correctly until the line:

The code executes normally until it reaches this line in MD_MAX72xx_buf.cpp

if (value & (1 << i))

This line gets hit, but immediately after, the execution jumps to the error.c file at the following snippet:

if (Error_policy_D == Error_SPIN) {

    for(;;) {

    }

}

It seems like some assertion or internal error handling is being triggered, but there is no clear error or assert message explaining the cause.

Kindly support to resolve the issue.

MajicDesigns, 2025-04-30:
Hi there from the 'team' of 1.

I don't have your processor so you will need to do some debugging on your system to see what the cause could be. You can turn on debugging code in the library by changing MAX_DEBUG to 1 in the file MD_MAX72xxlib.h. You will need to turn on the serial output stream in the main application to see debug output.

Without more information I can't see what could be throwing an exception in the code you highlighted.

MajicDesigns, 2025-07-22:
No response from OP

closed 2025-07-22 by MajicDesigns